autowarefoundation/agnocast #473: MultiThreadedAgnocastExecutorのMutuallyExclusive管理方法の問題点

atsushi421, 2025-03-05:
複数のagnocast callbackとros2 callbackが同一のMutuallyExclusive callback groupに所属しているとする。この時 `MultiThreadedAgnocastExecutor` では、agnocast用のスレッドとros2用のスレッドが以下のように `can_be_taken_from()` を取り合う。

- [agnocast](https://github.com/tier4/agnocast/blob/002812d92f003117aab15b4185fd23f8951d1650/src/agnocastlib/src/agnocast_executor.cpp#L169) 
- [ros2 (get_next_executable内)](https://github.com/tier4/agnocast/blob/002812d92f003117aab15b4185fd23f8951d1650/src/agnocastlib/src/agnocast_multi_threaded_executor.cpp#L82)

そのため、`can_be_taken_from()` を取れるかどうかは完全にタイミング次第であり、「agnocast callbackしか実行されない」・「ros2 callbackしか実行されない」といった現象が発生する可能性がある。実際に、統合テストで過負荷状態でないにも関わらず[この現象が見受けられた](https://star4.slack.com/archives/C07FL8616EM/p1741138164796809?thread_ts=1741138100.633099&cid=C07FL8616EM)。

Koichi98, 2025-03-05:
本件はデグレと言えると思います。

その上で、MultithreadedExecutorの実装が現在はagnocast_spin専用スレッドとros2_spin専用スレッドに完全に別れていますが、例えばメインスレッドはどちらも実行するように変更するのはどうでしょう？(SingleThreadedExecutorのように)

Koichi98, 2025-03-05:
もはや2スレッド以上いるときには現在のSingleThreadedExecutorとその逆の優先度でros2_spinとagnocast_spinを行うスレッドがいてもいいかもですね。
(現在はget_next_agnocast_executablesではnext_exec_timeout_msブロックし、ros2のget_next_executableはブロックしないのでその逆をするスレッドを作る。)

Koichi98, 2025-03-05:
なぜros2ではこのような問題が起きないのかを確認したのですが、ros2ではget_next_executableしてくる時点で先にcan_be_taken_fromを確認していてexecutable_any_executable時点ではtrueであることを前提にしていそうですかね？これは正しいのでしょうか？executableをとってくるときはtrueでもexecute時にはfalseになっている可能性があるように思えるのですが、どっかでロックとったりして防いでるんですかね？

atsushi421, 2025-03-05:
> 現実装のように一定周期で確認するよりはcan_be_takenがtrueになるタイミングを得られる可能性が高く

これで可能性が上がる理由が理解できていないので、もう少し説明お願いしたいです！

sykwer, 2025-03-05:
@atsushi421 
`can_be_taken_from` がfalseと最初に判断されたexecutableが、
- 今までの実装だと、一定周期のsleepからの起床時にしかチェックされない
- 改善案だと、全ての `get_next_agnocast_executables` が再実行のチャンスになる
からです

atsushi421, 2025-03-05:
なるほど、理解できました。

後出しで申し訳ないですが、統合テストでfailしているケースでは、全てros2 callbackの方が実行されないケースでした。

既存実装と比べて現状一番問題かなと思っているのは、agnocastのcallbackのqueueとros2のcallbackのqueueが別なので、例えばtimestamp=100のagnocast callbackとtimestamp=5のros2 callbackがcan_be_taken_fromの取得権を同等に持ってしまうという点です。既存実装では、[queueの先頭から見ている](https://github.com/ros2/rclcpp/blob/19773973a80d753c6fa028b0b548462fbbef122d/rclcpp/include/rclcpp/strategies/allocator_memory_strategy.hpp#L251)ので、古いメッセージが優先されるはずです。

sykwer, 2025-03-05:
ros2の `get_next_executables` で、 `can_be_taken_from_==false` と判定されて、実行待ちのexecutablesが何なのかが隠蔽されている (Agnocast側から知ることができない) というのが根本原因ですね。どうしたものか。。。

atsushi421, 2025-03-06:
https://star4.slack.com/archives/C07FL8616EM/p1741252636079389

atsushi421, 2025-03-12:
agnocast/ros2間でFIFO性を保証する実装は将来的にはやるべきなので、このissueは残しておく