autowarefoundation/agnocast #508: `AGNOCAST_SUBSCRIBER_ADD_CMD` sometimes fails because `copy_from_user` encounters an error

bdm-k, 2025-03-11:
issue の概要はタイトルの通り。

現時点で分かっていること

- e2e_test_2to2 実行時に発生する
- topic_name の `copy_from_user` がエラーになる
- エラーが発生する時、ユーザー側のソースメモリ領域は必ずページ境界をまたいでいる
- エラーが発生する時、ページ境界まではコピーできていて次のページの内容は全くコピーできていない

veqcc, 2025-03-11:
> エラーが発生する時、ユーザー側のソースメモリ領域は必ずページ境界をまたいでいる
エラーが発生する時、ページ境界まではコピーできていて次のページの内容は全くコピーできていない

I want you to check these through
- a small experimental code
- documentation

bdm-k, 2025-03-14:
I printed the address to the kernel log when `copy_from_user` failed and found that the memory region to be copied always spanned two pages.

I also printed the return value of the failed `copy_from_user` call, and the value was always the size from the address to the page boundary. Since the return value of `copy_from_user` indicates the size that could not be copied, we can say that the copy was successful up to the page boundary.

bdm-k, 2025-03-17:
I believe the error is caused by attempting to copy data from a page that does not exist in the first place. The changes made in  commit https://github.com/tier4/agnocast/commit/07929ffd5cb62746f034e83c61b2f996e3f4cab6 always causes this error and conceptually explains what I mean by "a page that does not exist in the first place."

PR #534 (2025-03-24): fix: pass string length along with its pointer
## Description
closes #508

This PR solves issue #508 by passing string length along with its pointer when invoking the ioctls.

### Key points of the implementation
A new struct `name_info` is introduced:
```cpp
struct name_info
{
  const char * ptr;
  uint64_t len;
};
```
This struct is used when passing a topic name or a node name to the kernel module.

## How was this PR tested?

- [x] Autoware (required)
- [x] `bash scripts/e2e_test_1to1_with_ros2sub` (required)
- [x] `bash scripts/e2e_test_2to2` (required)
- [x] sample application

## Notes  [... cut]
closed 2025-03-24 by bdm-k